PixelUpdater/PixelUpdater #2: Cannot search for OTA updates with IOException -> SocketException

alexwh, 2024-12-17:
I'm using a P9PXL with Android 15 AP4A.241205.013. If I uninstall the Magisk module, reboot and reinstall the Magisk module, the update check and installer works one time, but any subsequent attempt to check for updates fails with the below error:

`Failed to install OTA update, IOException (Failed to download update info) -> SocketException (Binding socket to network 123 failed: EPERM (Operation not permitted)) -> ErrnoException (Binding socket to network 123 failed: EPERM (Operation not permitted))`

Logs:
[magisk.log](https://github.com/user-attachments/files/18172400/magisk.log)
[install.log](https://github.com/user-attachments/files/18172401/install.log)
[check.log](https://github.com/user-attachments/files/18172402/check.log)
[pixelupdater_selinux.log](https://github.com/user-attachments/files/18172404/pixelupdater_selinux.log)

badabing2005, 2024-12-19:
After you uninstall and reboot, can you please run this command
```
ls -lR /storage/emulated/0/Android/data/com.github.pixelupdater.pixelupdater
```

if it finds anything please run this command
```
rm -rf /storage/emulated/0/Android/data/com.github.pixelupdater.pixelupdater
```

Then try to install and see if it works.

alexwh, 2024-12-19:
It returned:

```
❯ adb shell                                                                              
* daemon not running; starting now at tcp:5037
* daemon started successfully
komodo:/ $ ls -lR /storage/emulated/0/Android/data/com.github.pixelupdater.pixelupdater
/storage/emulated/0/Android/data/com.github.pixelupdater.pixelupdater:
total 3
drwxrws--- 2 u0_a48 ext_data_rw 3452 2024-12-14 21:36 files

/storage/emulated/0/Android/data/com.github.pixelupdater.pixelupdater/files:
total 20
-rw-rw---- 1 u0_a48 ext_data_rw 5419 2024-12-19 14:13 check.log
-rw-rw---- 1 u0_a48 ext_data_rw 5933 2024-12-14 21:36 install.log
-rw-rw---- 1 u0_a48 ext_data_rw 2099 2024-12-14 21:41 magisk.log
komodo:/ $
```

They are:
[check.log](https://github.com/user-attachments/files/18199057/check.log)
[install.log](https://github.com/user-attachments/files/18199056/install.log)
[magisk.log](https://github.com/user-attachments/files/18199054/magisk.log)

After removing them and reinstalling/rebooting, I was able to check for updates 3 times, then I toggled debug mode and the same error occurred. Post reinstall logs:

[pixelupdater_selinux.log](https://github.com/user-attachments/files/18199112/pixelupdater_selinux.log)
[check.log](https://github.com/user-attachments/files/18199117/check.log)
(magisk.log and install.log didn't exist)

badabing2005, 2025-01-17:
@alexwh 
See if v0.108 works for you.

Thanks

alexwh, 2025-01-17:
Same as last time - I updated to v0.108, rebooted, used it to update to AP4A.250105.002 - it said couldn't patch magisk but clicking retry seemed to work, and then after rebooting again the update check worked once or twice then back to socket EPERM.

[magisk.log](https://github.com/user-attachments/files/18447413/magisk.log)
[install.log](https://github.com/user-attachments/files/18447414/install.log)
[check.log](https://github.com/user-attachments/files/18447415/check.log)

There was no pixelupdater_selinux.log.

badabing2005, 2025-01-17:
Thanks for the prompt testing and reply.
> it said couldn't patch magisk but clicking retry seemed to work
This is a known issue that needs to be addressed, Magisk is returning an errorcode even though the patch is successful.
If you reboot without retry, you'd see that it had worked.
We'll have this fixed, but it is harmless.

> and then after rebooting again the update check worked once or twice then back to socket EPERM
This one I cannot reporduce.
To be clear, all you're doing is check for updates?
I did it like 20 times and it still worked.
You don't use some firewall app / tool , do you?

alexwh, 2025-01-17:
The only thing I can think of it being is AdAway or maybe Wireguard, both installed with root, but the Wireguard tunnel is local only (not redirecting all traffic).

badabing2005, 2025-01-17:
I use Adaway and Wireguard, though Wireguard is not active when testing, and mine is not setup for local only.
Can you give it a shot by stopping it and see if it makes a difference?

badabing2005, 2025-01-17:
How do you add PixelUpdate to the excluded list, I don't see such option.
Yes I am using Kernel module mode.

Why is you Interface subnet mas /24?
Shouldn't it be /32?

alexwh, 2025-01-17:
Hit "All Applications" under the DNS servers textbox.
It's a /24 because I have the tunnel active at all times and use it to connect to other hosts in the network, not to tunnel internet traffic.

badabing2005, 2025-01-17:
Thanks for pointing out where the Application exclusion is.

You probably know a lot more about Wireguard than I do, but with my limited knowledge, I think

AllowedIPs = 10.0.0.0/24
This tells your device that any traffic destined for an IP within the 10.0.0.0/24 subnet (which includes 10.0.0.0 through 10.0.0.255) should be sent through the WireGuard VPN.

Whereas this 
Address = 10.0.0.3/24
controls what IPs it is listening on.

Address = 10.0.0.3/32
The /32 subnet mask just tells the device that it is only using this single IP (10.0.0.3), not the entire 10.0.0.0/24 subnet.
However, this shouldn't prevent communication with other devices within that subnet.

badabing2005, 2025-01-17:
OK thanks for the update.

alexwh, 2025-03-17:
Same issue as before if the app isn't excluded.

closed 2025-07-31 by badabing2005