PixelUpdater/PixelUpdater #3: Crashes on update check, even though it has a super root

ostrolucky, 2024-12-29:
As you can see on this video, pixel updater is getting granted superuser access
[screen-20241229-233642.mkv.zip](https://github.com/user-attachments/files/18271137/screen-20241229-233642.mkv.zip)
but when clicking update, application crashes. Adb logcat reveals this error:
```
12-29 23:32:40.025 16218 17854 E AndroidRuntime: FATAL EXCEPTION: Thread-5
12-29 23:32:40.025 16218 17854 E AndroidRuntime: Process: com.github.pixelupdater.pixelupdater, PID: 16218
12-29 23:32:40.025 16218 17854 E AndroidRuntime: java.io.IOException: Cannot run program "logcat": /storage/emulated/0/Android/data/com.github.pixelupdater.pixelupdater/files/check.log: open failed: EACCES (Permission denied)
12-29 23:32:40.025 16218 17854 E AndroidRuntime: 	at java.lang.ProcessBuilder.start(ProcessBuilder.java:1050)
12-29 23:32:40.025 16218 17854 E AndroidRuntime: 	at com.github.pixelupdater.pixelupdater.updater.UpdaterThread.startLogcat(Unknown Source:64)
12-29 23:32:40.025 16218 17854 E AndroidRuntime: 	at com.github.pixelupdater.pixelupdater.updater.UpdaterThread.run(Unknown Source:14)
12-29 23:32:40.025 16218 17854 E AndroidRuntime: Caused by: java.io.FileNotFoundException: /storage/emulated/0/Android/data/com.github.pixelupdater.pixelupdater/files/check.log: open failed: EACCES (Permission denied)
12-29 23:32:40.025 16218 17854 E AndroidRuntime: 	at libcore.io.IoBridge.open(IoBridge.java:574)
12-29 23:32:40.025 16218 17854 E AndroidRuntime: 	at java.io.FileOutputStream.<init>(FileOutputStream.java:259)
12-29 23:32:40.025 16218 17854 E AndroidRuntime: 	at java.lang.ProcessImpl.start(ProcessImpl.java:118)
12-29 23:32:40.025 16218 17854 E AndroidRuntime: 	at java.lang.ProcessBuilder.start(ProcessBuilder.java:1029)
12-29 23:32:40.025 16218 17854 E AndroidRuntime: 	... 2 more
12-29 23:32:40.025 16218 17854 E AndroidRuntime: Caused by: android.system.ErrnoException: open failed: EACCES (Permission denied)
12-29 23:32:40.025 16218 17854 E AndroidRuntime: 	at libcore.io.Linux.open(Native Method)
12-29 23:32:40.025 16218 17854 E AndroidRuntime: 	at libcore.io.ForwardingOs.open(ForwardingOs.java:563)
12-29 23:32:40.025 16218 17854 E AndroidRuntime: 	at libcore.io.BlockGuardOs.open(BlockGuardOs.java:274)
12-29 23:32:40.025 16218 17854 E AndroidRuntime: 	at libcore.io.ForwardingOs.open(ForwardingOs.java:563)
12-29 23:32:40.025 16218 17854 E AndroidRuntime: 	at android.app.ActivityThread$AndroidOs.open(ActivityThread.java:8673)
12-29 23:32:40.025 16218 17854 E AndroidRuntime: 	at libcore.io.IoBridge.open(IoBridge.java:560)
12-29 23:32:40.025 16218 17854 E AndroidRuntime: 	... 5 more
12-29 23:32:40.028 16218 17854 E MainApplication: Saving logcat to /storage/emulated/0/Android/data/com.github.pixelupdater.pixelupdater/files/crash.log due to uncaught exception in Thread[Thread-5,5,main]
12-29 23:32:40.028 16218 17854 E MainApplication: java.io.IOException: Cannot run program "logcat": /storage/emulated/0/Android/data/com.github.pixelupdater.pixelupdater/files/check.log: open failed: EACCES (Permission denied)
12-29 23:32:40.028 16218 17854 E MainApplication: 	at java.lang.ProcessBuilder.start(ProcessBuilder.java:1050)
12-29 23:32:40.028 16218 17854 E MainApplication: 	at com.github.pixelupdater.pixelupdater.updater.UpdaterThread.startLogcat(Unknown Source:64)
12-29 23:32:40.028 16218 17854 E MainApplication: 	at com.github.pixelupdater.pixelupdater.updater.UpdaterThread.run(Unknown Source:14)
12-29 23:32:40.028 16218 17854 E MainApplication: Caused by: java.io.FileNotFoundException: /storage/emulated/0/Android/data/com.github.pixelupdater.pixelupdater/files/check.log: open failed: EACCES (Permission denied)
12-29 23:32:40.028 16218 17854 E MainApplication: 	at libcore.io.IoBridge.open(IoBridge.java:574)
12-29 23:32:40.028 16218 17854 E MainApplication: 	at java.io.FileOutputStream.<init>(FileOutputStream.java:259)
12-29 23:32:40.028 16218 17854 E MainApplication: 	at java.lang.ProcessImpl.start(ProcessImpl.java:118)
12-29 23:32:40.028 16218 17854 E MainApplication: 	at java.lang.ProcessBuilder.start(ProcessBuilder.java:1029)
12-29 23:32:40.028 16218 17854 E MainApplication: 	... 2 more
12-29 23:32:40.028 16218 17854 E MainApplication: Caused by: android.system.ErrnoException: open failed: EACCES (Permission denied)
12-29 23:32:40.028 16218 17854 E MainApplication: 	at libcore.io.Linux.open(Native Method)
12-29 23:32:40.028 16218 17854 E MainApplication: 	at libcore.io.ForwardingOs.open(ForwardingOs.java:563)
12-29 23:32:40.028 16218 17854 E MainApplication: 	at libcore.io.BlockGuardOs.open(BlockGuardOs.java:274)
12-29 23:32:40.028 16218 17854 E MainApplication: 	at libcore.io.ForwardingOs.open(ForwardingOs.java:563)
12-29 23:32:40.028 16218 17854 E MainApplication: 	at android.app.ActivityThread$AndroidOs.open(ActivityThread.java:8673)
12-29 23:32:40.028 16218 17854 E MainApplication: 	at libcore.io.IoBridge.open(IoBridge.java:560)
12-29 23:32:40.028 16218 17854 E MainApplication: 	... 5 more
12-29 23:32:40.617   551   551 E BpTransactionCompletedListener: Failed to transact (-32)
12-29 23:32:40.739  4179  4275 E native  : E0000 00:00:1735511560.729032    4275 mmap.cc:234] Error during Unmap / munmap: Invalid argument
12-29 23:32:41.456  1099  4017 E IPCThreadState: binder thread pool (1 threads) starved for 162 ms
12-29 23:32:41.818  3453  3623 E EuiccGoogle: [52] EuiccUpdater.startOtaIfNecessary: Timeout for waiting OTA complete in EuiccSupportPixel
```

I have a Pixel 8

badabing2005, 2024-12-30:
Thanks for reporting.
Can you please share the output of running the following commands on the phone.
```
$ su
# find /data/dalvik-cache -iname '*pixelupdater*'
# find /data/system/package_cache -iname '*pixelupdater*'
# pm list packages com.github.pixelupdater.pixelupdater
# ls -lah /data/ota_package
# ls -lR /storage/emulated/0/Android/data/com.github.pixelupdater.pixelupdater
```

Thanks

badabing2005, 2024-12-30:
Here's the issue, notice how there's different ownership of the files / directories.
This happens because of multiple installs, or going unrooted and then going back rooted.
[image]

For whatever reasons, the Magisk uninstall is not properly deleting 
`/storage/emulated/0/Android/data/com.github.pixelupdater.pixelupdater`
even though it's in the uninstall script.

You can try deleting that as root
```
rm -rf /storage/emulated/0/Android/data/com.github.pixelupdater.pixelupdater`
```
if that doesn't work, uninstall, reboot, check all of the above, make sure non are left, and then reinstall.
I'd have to figure out why Magisk is not uninstalling properly.

ostrolucky, 2024-12-30:
Indeed, removing that directory fixed the issue. So for me issue was solved, but looks like there might be a need for some fix in magisk uninstall script or so? Either way, I'll let you decide if you want to keep the issue open or close it. Thanks!

badabing2005, 2024-12-30:
Thanks for testing and reporting.
I'll keep it open unless I find no way around Magisk uninstall.

ostrolucky, 2024-12-30:
Would it be possible pixel updater checks (and fixes) these permissions on startup instead of assuming uninstall script always runs?

badabing2005, 2024-12-30:
That's a good idea, I'll look into it.

commit 6dba7e21 (2025-01-17): Fix #3 Delete a previous PixelFlasher data on install, as Magisk is not deleting it completely on uninstall.
closed 2025-01-17 by badabing2005