PixelUpdater/PixelUpdater #4: Verifying OTA update stuck

microlancer, 2025-01-06:
I checked the logs:

```
01-06 13:34:17.358  8411  8428 D UpdaterThread: onStatusUpdate(4/VERIFYING, 96.11846%)
01-06 13:34:17.358  8411  8411 D UpdaterService: Updating foreground notification for state: ProgressState(type=VERIFY, current=96, max=100)
01-06 13:34:18.044  8411  8428 D UpdaterThread: onStatusUpdate(4/VERIFYING, 97.11994%)
01-06 13:34:18.045  8411  8411 D UpdaterService: Updating foreground notification for state: ProgressState(type=VERIFY, current=97, max=100)
01-06 13:34:18.363  8411  8428 D UpdaterThread: onStatusUpdate(4/VERIFYING, 98.12116%)
01-06 13:34:18.364  8411  8411 D UpdaterService: Updating foreground notification for state: ProgressState(type=VERIFY, current=98, max=100)
01-06 13:34:18.982  8411  8428 D UpdaterThread: onStatusUpdate(4/VERIFYING, 99.12258%)
01-06 13:34:18.983  8411  8411 D UpdaterService: Updating foreground notification for state: ProgressState(type=VERIFY, current=99, max=100)
01-06 13:34:19.131  8411  8428 D UpdaterThread: onStatusUpdate(4/VERIFYING, 100.0%)
01-06 13:34:19.132  8411  8411 D UpdaterService: Updating foreground notification for state: ProgressState(type=VERIFY, current=100, max=100)
01-06 13:34:20.204  8411  8428 D UpdaterThread: onStatusUpdate(4/VERIFYING, 100.0%)
01-06 13:34:20.205  8411  8411 D UpdaterService: Updating foreground notification for state: ProgressState(type=VERIFY, current=100, max=100)
01-06 13:34:20.206  8411  8428 D UpdaterThread: onStatusUpdate(5/FINALIZING, 0.0%)
01-06 13:34:20.217  8411  8411 D UpdaterService: Updating foreground notification for state: ProgressState(type=FINALIZE, current=0, max=100)
01-06 13:38:39.409  8411  8411 D VRI[SettingsActivity]: visibilityChanged oldVisibility=true newVisibility=false
```

Maybe it's a bug somewhere because it goes from 100 to suddenly 0. But on the phone, it shows the notification with the progress bar for "Finalizing OTA update" at 0 and the options show "Pause" or "Cancel". I can't tell if it's done, or if it's broken.

closed 2025-01-06 by microlancer

simplydat, 2025-01-09:
@microlancer Why did you close this issue? Did you solve it? I just ran into the same issue updating my pixel 8 pro to Jan 2025.

badabing2005, 2025-01-10:
@simplydat 
Lately I've seen `failed to Magisk patch inactive slot` being a benign message.
If you click on the retry button after seeing that, and it reports success, it is a benign issue, in fact it had already worked before the retry.
However if you see that it fails again, then it's a problem.
Logs would help.

I have not seen / experienced the issue about disk consumption.
If you're rooted, you can check where the space is consumed.

We can consider adding an option to Clear OTA data, if that happens to be the issue.

In the meantime, if I get some time, I'll put together a quick Magisk module where the action button would do that.

badabing2005, 2025-01-11:
Here's a Magisk module that clears pending / in-progress updates.
[clear_ota_updates.zip](https://github.com/user-attachments/files/18383463/clear_ota_updates.zip)

microlancer, 2025-01-15:
I closed it because I think it was user error. I had mixed up my versions of Magisk, and tried to install a version that didn't match the one that was installed on the init image. I gave up trying to rescue my rooted state and just grabbed the normal OTA which would cause me to lose root, but I then used Magisk to create a new init image, flashed it, and restarted, and everything was working cleanly again.

I learned that Magisk is sensitive about which version of Magisk is installed as an app, and which version of Magisk is installed together with the flashed init -- and these should probably match 100% or else gonna have a hard time like I did.

simplydat, 2025-01-17:
@badabing2005 Thank you. Can confirm after waiting long enough for the initial failed message, rebooting the phone still does apply the update. Maybe it's worth investigating why there is that "benign message". Thank you.

@microlancer Thanks for the update.

badabing2005, 2025-01-17:
Have to hunt down why Magisk is returning an returncode 1 when things go well, possibly Magisk issue or our use of Magisk.
Short of that, have to put in another detection mechanism to determine if it was successful and not rely on the returncode.