PixelUpdater/PixelUpdater #5: Version for APatch

Nihoru, 2025-01-08:
Will a version for APatch be developed?
At the moment this module does not work with it, because APatch patches boot.img, not init_boot.img

badabing2005, 2025-01-08:
Probably not anytime soon.
There are too many variations, then of course there are the forks (the Next flavors), and not to forget KernelSU flavors as well.
Can it be done? of course.
But it takes time and a lot of testing.

Nihoru, 2025-01-08:
Got it, thanks for the reply

badabing2005, 2025-05-09:
See this [post](https://xdaforums.com/t/has-anyone-used-the-pixel-updater-tool-on-github-for-monthly-updates.4698501/post-90090401).
It's not all seamless as with Magisk, but it allows you to update (not reboot) and patch inactive slot.