PixelUpdater/PixelUpdater #6: KernelSU updating works already

Xarber, 2025-01-24:
This is not really an issue, more a suggestion to remove limitations to kernelsu users:
I just installed the A16 Beta 1, updating from A16 DP2, and managed to keep my kernelsu installation without any problems, but I did have to perform these steps 

0. Restore the stock images from KernelSU (To ensure the OTA update is successful)
1. Enable "Magisk patch installed update" (this is needed to remove the "mismatched magisk settings" notification)
2. Look for ota updates
3. Disable the "Magisk patch installed update"
4. Click on the install button on the notification with the available OTAs
5. After it's done, go back on kernelSU and "Install to the inactive slot"
6. Reboot the device and it should be updated with kernelsu installed instead of magisk.

My suggestion is just to add an option to "Force allow mismatched magisk settings", maybe in advanced settings, so that it's easier for the user to install an OTA and keep kernelsu without a PC.
(kernelSU can't be installed if the ota boot file is already patched with magisk)

badabing2005, 2025-01-24:
That's the manual of doing things and has nothing to do with PixelUpdater.
PixelUpdater does not need to do any of the steps described in steps 0, 1, 3, .. it updates even if your device is patched and then applies the patch after the update.
Currently it only supports Magisk, but KernelSU and Apatch support is being considered.

atikbinmasud, 2025-02-02:
I'd be great if KSU gets the support for it.

badabing2005, 2025-05-09:
See this [post](https://xdaforums.com/t/has-anyone-used-the-pixel-updater-tool-on-github-for-monthly-updates.4698501/post-90090401).
It's not all seamless as with Magisk, but it allows you to update (not reboot) and patch inactive slot.

wechselgeld, 2025-05-11:
Yes, that would be great! I switched to KSU, because of the banking apps I use — they will detect Magisk, no matter what I do, KSU works just fine. I would help, but I can't code in the languages used :/
Keep the good work up! The updater looks & works good!