PixelUpdater/PixelUpdater #8: Pixel 6 Error while patching OTA

Dart2112, 2025-02-01:
See the below log
It seems like the patching worked but it's looking for preinit which I don't think was needed until Pixel 7 based on my experience of rooting my P6 and my partners P7

If you need the other logs, please let me know
```
- Unpacking boot image
Parsing boot image: [/dev/block/by-name/boot_b]
HEADER_VER      [4]
KERNEL_SZ       [24983196]
RAMDISK_SZ      [0]
PAGESIZE        [4096]
CMDLINE         []
KERNEL_FMT      [lz4_legacy]
unexpected ASN.1 DER tag: expected SEQUENCE, got APPLICATION [1] (primitive)
VBMETA
- Checking ramdisk status
- Stock boot image detected
- Patching ramdisk
- Pre-init storage partition: sda8
Add file [init] (100750)
Create directory [overlay.d] (0750)
Create directory [overlay.d/sbin] (0750)
Add file [overlay.d/sbin/magisk.xz] (100644)
Add file [overlay.d/sbin/stub.xz] (100644)
Add file [overlay.d/sbin/init-ld.xz] (100644)
Patch with flag KEEPVERITY=[true] KEEPFORCEENCRYPT=[true]
Create directory [.backup] (0000)
Add file [.backup/.magisk] (100000)
Dumping cpio: [ramdisk.cpio]
- Repacking boot image
Parsing boot image: [/dev/block/by-name/boot_b]
HEADER_VER      [4]
KERNEL_SZ       [24983196]
RAMDISK_SZ      [0]
PAGESIZE        [4096]
CMDLINE         []
KERNEL_FMT      [lz4_legacy]
unexpected ASN.1 DER tag: expected SEQUENCE, got APPLICATION [1] (primitive)
VBMETA
Repack to boot image: [new-boot.img]
RAMDISK_FMT: [raw] -> [lz4_legacy]
HEADER_VER      [4]
KERNEL_SZ       [24983196]
RAMDISK_SZ      [270458]
PAGESIZE        [4096]
CMDLINE         []
- Flashing new boot image
Cleaning up...
- Unable to find preinit dir
- 
```

Dart2112, 2025-02-01:
Just rebooted. The update actually did install successfully. But I had a notification claiming the update had failed in the patching part. But after reboot I was updated and retained root.
Just the notification needs to be fixed I guess.

badabing2005, 2025-03-17:
@Dart2112 please test it with [this](https://github.com/PixelUpdater/PixelUpdater/releases/tag/v0.109) version, I believe it should have resolve the issue.

closed 2025-03-19 by badabing2005