PixelUpdater/PixelUpdater #10: Translation of the application into Russian

Xacker404, 2025-02-23:
I have translated the application into Russian, you can apply my file for the next updates.

[values.zip](https://github.com/user-attachments/files/18933385/values.zip)

Xusysy, 2025-03-17:
[image]
[image]
There is no translation into Russian:(

vladrevers, 2025-03-17:
@badabing2005
1) Replace vi with ru: https://github.com/PixelUpdater/PixelUpdater/blob/aad545f8941ada92b9ee63869b58c076b8ec4d96/app/src/main/res/xml/locales_config.xml#L8
2) add "ru" https://github.com/PixelUpdater/PixelUpdater/blob/880fbb9c35a99a27e4ffd323a6a7d9f22eed6aa2/app/build.gradle.kts#L165
3) delete "not translatable string "in translate:
https://github.com/PixelUpdater/PixelUpdater/blob/880fbb9c35a99a27e4ffd323a6a7d9f22eed6aa2/app/src/main/res/values-ru/strings.xml#L9
https://github.com/PixelUpdater/PixelUpdater/blob/880fbb9c35a99a27e4ffd323a6a7d9f22eed6aa2/app/src/main/res/values-ru/strings.xml#L16
4) plurals ru need `many` and `few` variant
```xml
<resources>
	<plurals name="mtrl_badge_content_description">
		<item quantity="one">%d новое уведомление</item>
		<item quantity="many">%d новых уведомлений</item>
		<item quantity="few">%d новых уведомлений</item>
	</plurals>
</resources>
```

badabing2005, 2025-03-17:
[Completed](https://github.com/PixelUpdater/PixelUpdater/releases/tag/v0.109).

closed 2025-03-17 by badabing2005

Xusysy, 2025-03-17:
Super! Everything is working fine :)