PixelUpdater/PixelUpdater #13: Possible unused files in APK that increase its size

vladrevers, 2025-03-18:
The compiled APK contains files that may not be used and could be removed to optimize the size:

1. `assets/archive.tar` (1.14MB) - contains the repository source code
2. Directory `org/bouncycastle` (1.34MB) - files for a library that might not be used

In the original project, bouncycastle was explicitly used only in the UpdaterThread.kt file, and it's unclear if it's necessary in the current version of the project.

Also found resources that don't significantly affect the APK size but aren't used in the application and create unnecessary files in the project:
- `res/drawable/ic_notification.xml` (likely remained from the original project, currently `ic_notifications.xml` is used instead)
- `res/drawable/ic_launcher_background.xml` (currently ic_launcher.xml uses `@color/ic_launcher_background` instead)
- `res/mipmap-anydpi-v26` (files identical to mipmap-anydpi, and with `minSdkVersion=33` there's no need for separation)
- folder `res/mipmap-anydpi/ic_launcher_round.xml` (not specified in the manifest and redundant for Android 7+)
- `res/values-*/plurals.xml` (contains one value that isn't used anywhere)

I think we should consider removing these files/resources if they're actually not needed for the application to work.

badabing2005, 2025-03-18:
Where are you seeing these?
Which branch?
- `assets/archive.tar` (1.14MB) - contains the repository source code
- Directory `org/bouncycastle` (1.34MB) - files for a library that might not be used

- `app\src\main\res\drawable\ic_notification.xml` is used for notification icons 
- `app\src\main\res\drawable\ic_launcher_background.xml` is part of the app's launcher icon definition
- `folder rapp\src\main\es/mipmap-anydpi-v26` contains adaptive icon definitions for Android 8.0+
- `app\src\main\app\src\main\res/mipmap-anydpi/ic_launcher_round.xml` defines the round version of the launcher icon

vladrevers, 2025-03-18:
@badabing2005
"Where are you seeing these?"
I can definitively confirm these files are in the compiled PixelUpdater.apk when opened with an archive viewer.

The archive.tar file is located in assets/ and is not used anywhere in the application code.

In the original source code, it looks like it's being included during build:
https://github.com/PixelUpdater/PixelUpdater/blob/ded0f3937ca1c73d98400503dfe786cb6cd8fdf8/app/build.gradle.kts#L260
https://github.com/PixelUpdater/PixelUpdater/blob/ded0f3937ca1c73d98400503dfe786cb6cd8fdf8/app/build.gradle.kts#L295

This would likely occur in any branch during the build process.

vladrevers, 2025-03-18:
> These are fine

> app\src\main\res\drawable\ic_notification.xml is used for notification icons

The notification icon is using ic_notification.xml, but there's also ic_notificationS.xml nearby which isn't being used.

> app\src\main\res\drawable\ic_launcher_background.xml is part of the app's launcher icon definition

ic_launcher_background.xml was used previously, but now there's a reference to a color in the color resources, and the background file in drawable is no longer used.
https://github.com/PixelUpdater/PixelUpdater/blob/f376788d91126632b21b64eaea62fba5f94d475d/app/src/main/res/mipmap-anydpi/ic_launcher.xml#L3

> folder rapp\src\main\es/mipmap-anydpi-v26 contains adaptive icon definitions for Android 8.0+

Currently the app has both mipmap-anydpi-v26 and mipmap-anydpi. The mipmap-anydpi folder isn't used at all, because with -v26 present, it would only be used on SDK 25 and below, which is irrelevant since the app has a minimum SDK of 33 (Android 13). Since the files in mipmap-anydpi and mipmap-anydpi-v26 are identical, we could simply delete mipmap-anydpi-v26. If they were different, we could delete mipmap-anydpi and rename mipmap-anydpi-v26 by removing the -v26 suffix.

> app\src\main\app\src\main\res/mipmap-anydpi/ic_launcher_round.xml defines the round version of the launcher icon

ic_launcher_round.xml isn't currently used. For it to be used, AndroidManifest.xml would need to include android:roundIcon="@mipmap/ic_launcher_round", but it doesn't have this and doesn't need it. roundIcon is only needed for Android 7, because Google initially added support for round icons, but in Android 8+ the system makes the necessary shape by cropping the background from ic_launcher (typically circular, or rounded squares on Samsung/Xiaomi etc.).

badabing2005, 2025-03-18:
> I'm seeing these in the compiled PixelUpdater.apk when opened with an archive viewer.
This is actually intentional functionality, not a mistake.
License compliance GPL-3.0 requires source availability.

> bouncycastle
I'm not touching that for now without extensive testing.

For the other suggestions, I pushed an update, take a look.

badabing2005, 2025-03-18:
completed in 0.109.1

closed 2025-03-18 by badabing2005

vladkryv, 2025-03-18:
> I'm seeing these in the compiled PixelUpdater.apk when opened with an archive viewer.
> This is actually intentional functionality, not a mistake.
> License compliance GPL-3.0 requires source availability.

Thank you for clarifying about archive.tar. If the code that GitHub attaches to releases isn't sufficient for GPL-3.0 compliance, perhaps the archive.tar could be placed in the Magisk module zip archive instead of the APK, and possibly be removed during installation in customize.sh :D

> bouncycastle
> I'm not touching that for now without extensive testing.

Yes, bouncycastle would need verification. I tried building after removing the dependency lines and it compiled normally, but it still might affect something.

> For the other suggestions, I pushed an update, take a look.

Great! Everything looks good now, only plurals.xml remains.

vladrevers, 2025-03-18:
> I'm seeing these in the compiled PixelUpdater.apk when opened with an archive viewer.
> This is actually intentional functionality, not a mistake.
> License compliance GPL-3.0 requires source availability.

Thank you for clarifying about archive.tar. If the code that GitHub attaches to releases isn't sufficient for GPL-3.0 compliance, perhaps the archive.tar could be placed in the Magisk module zip archive instead of the APK, and possibly be removed during installation in customize.sh :D

> bouncycastle
> I'm not touching that for now without extensive testing.

Yes, bouncycastle would need verification. I tried building after removing the dependency lines and it compiled normally, but it still might affect something.

> For the other suggestions, I pushed an update, take a look.

Great! Everything looks good now, only plurals.xml remains.