PixelUpdater/PixelUpdater #15: "Bootloader status" locked and "Magisk will not be installed after an OTA"

CrossiDev-Studio, 2025-03-28:
In Pixelflasher it shows me, that the bootloader status is "locked" and also the hint "Magisk will not be installed after an OTA" - is this normal?

CrossiDev-Studio, 2025-03-28:
Oh yeah sorry, I mean PixelUpdater.

See screenshot

[]()[image]

badabing2005, 2025-03-28:
And you've granted root rights to PixelUpdater?
Can you take a screenshot of the issue?

CrossiDev-Studio, 2025-03-28:
Yes I did and restarted but still 

[image]

badabing2005, 2025-03-28:
Scroll down and enable this.

[image]
It's in the debug settings, you must have set it off.

closed 2025-03-28 by badabing2005