PixelUpdater/PixelUpdater #16: Update error

AstolfoBestGirl1, 2025-04-03:
Pixel Updater failed to install OTA update IOException (Unexpected size of central directory (1197))

Komodo BP22.250221.013
trying to update to BP22.250221.015 via url

badabing2005, 2025-04-03:
Sorry for being blunt, do you expect any solution to a problem for which you don't provide any detlais, no logs, nothing about the URL you pasted into the settings, did it get to the point of finding the update package or did this happen during the installation of the update?

I won't close this ticket immediately, but without your help, I can't help you.

AstolfoBestGirl1, 2025-04-04:
URL: https://dl.google.com/developers/android/baklava/images/factory/komodo_beta-bp22.250221.015-factory-b504db81.zip

where can i get logs of this app?

badabing2005, 2025-04-04:
Enable [debug](https://github.com/PixelUpdater/PixelUpdater?tab=readme-ov-file#debug-mode), and you'd see the option.

Also
> did it get to the point of finding the update package or did this happen during the installation of the update?

AstolfoBestGirl1, 2025-04-04:
--------- beginning of main
04-04 22:51:53.540 17725 17730 W er.pixelupdater: Suspending all threads took: 5.506ms
04-04 22:51:53.541 17725 17730 W er.pixelupdater: Cleared Reference was only reachable from finalizer (only reported once)
04-04 22:51:53.545 17725 17730 I er.pixelupdater: Background concurrent mark compact GC freed 2619KB AllocSpace bytes, 7(140KB) LOS objects, 91% free, 2297KB/26MB, paused 6.041ms,2.737ms total 43.646ms
04-04 22:51:53.545 17725 17730 W er.pixelupdater: Reducing the number of considered missed Gc histogram windows from 1383 to 100
04-04 22:51:53.561 17725 17725 D UpdaterService: Received intent: Intent { act=com.github.pixelupdater.pixelupdater.updater.UpdaterService.start xflg=0x4 cmp=com.github.pixelupdater.pixelupdater/.updater.UpdaterService (has extras) }
04-04 22:51:53.561 17725 17725 D UpdaterService: Updating foreground notification for state: null
04-04 22:51:53.566 17725 17725 D Notifications: Using provided icon: 2131230875
04-04 22:51:53.569 17725 17725 D UpdaterService: Creating new updater thread
04-04 22:51:53.570 17725 17725 D UpdaterService: Updating foreground notification for state: null
04-04 22:51:53.570 17725 17725 D Notifications: Using provided icon: 2131230875
04-04 22:51:53.610 17725 32189 D UpdaterThread: onStatusUpdate(0/IDLE, 0.0%)
04-04 22:51:53.613 17725 17725 D UpdaterService: More silent behavior for periodic job: true
04-04 22:51:53.614 17725 17725 D UpdaterService: Trying to stop service if possible
04-04 22:51:53.614 17725 32198 D UpdaterThread: Starting log file (v0.109.1)
04-04 22:51:53.627 17725 32198 D UpdaterThread: Waiting for initial engine status
04-04 22:51:53.628 17725 17725 D UpdaterService: Updating foreground notification for state: ProgressState(type=INIT, current=0, max=0)
04-04 22:51:53.628 17725 17725 D Notifications: Using provided icon: 2131230875
04-04 22:51:53.629 17725 32198 D UpdaterThread: Initial status: 0/IDLE
04-04 22:51:53.629 17725 32198 D UpdaterThread: Starting new update because engine is idle
04-04 22:51:53.633 17725 17725 D UpdaterService: Updating foreground notification for state: ProgressState(type=CHECK, current=0, max=0)
04-04 22:51:53.633 17725 17725 D Notifications: Using provided icon: 2131230875
04-04 22:51:53.664 17725 32198 D UpdaterThread: OTA URL: https://dl.google.com/developers/android/baklava/images/factory/komodo_beta-bp22.250221.015-factory-b504db81.zip
04-04 22:51:54.486 17725 32198 E UpdaterThread: Failed to install update
04-04 22:51:54.486 17725 32198 E UpdaterThread: java.io.IOException: Unexpected size of central directory (1197)
04-04 22:51:54.486 17725 32198 E UpdaterThread: 	at com.github.pixelupdater.pixelupdater.updater.UpdaterThread.checkForUpdates(Unknown Source:1291)
04-04 22:51:54.486 17725 32198 E UpdaterThread: 	at com.github.pixelupdater.pixelupdater.updater.UpdaterThread.run(Unknown Source:734)
04-04 22:51:54.486 17725 17725 D Notifications: Using provided icon: 2131230875
04-04 22:51:54.487 17725 17725 D Notifications: notification 2: Failed to install OTA update, IOException (Unexpected size of central directory (1197))
04-04 22:51:54.488 17725 32198 D UpdaterThread: Stopping log file
04-04 22:51:54.492 17725 17725 D UpdaterService: Trying to stop service if possible
04-04 22:51:54.492 17725 17725 D UpdaterService: Updater thread completed; stopping service

badabing2005, 2025-04-04:
Seriously man, there is a create support file button, why don't you use that and attach?
there are mnore files than this.
and please do it immediately after you get the error, logcat buffer is small and if you wait, you'd be providing logcat that does not include the relevant failure logs.

AstolfoBestGirl1, 2025-04-05:
[pu_support_2025-04-05_08-27-41.zip](https://github.com/user-attachments/files/19613638/pu_support_2025-04-05_08-27-41.zip)

badabing2005, 2025-04-05:
`https://dl.google.com/developers/android/baklava/images/factory/komodo_beta-bp22.250221.015-factory-b504db81.zip`

This is a factory image not an OTA image
This is the correct URL
https://dl.google.com/developers/android/baklava/images/ota/komodo_beta-ota-bp22.250221.015-69ceb9a4.zip

AstolfoBestGirl1, 2025-04-05:
Thanks! It worked!

closed 2025-04-05 by AstolfoBestGirl1