PixelUpdater/PixelUpdater #20: Support for updating Play System Updates

RareMaker50, 2025-04-17:
Stock play system updates and OEM ota updates don't work with root, if possible introducing a way to force/manually update Google play system updates, and Google play store/service apps without needing to unroot and wait until you receive the rollout.

badabing2005, 2025-04-18:
Sorry but I don't understand the issue.
This program's scope is to do monthly OS update on rooted Pixels.
It has nothing to do with Google Play Store or Services.

RareMaker50, 2025-04-18:
Not an issue, just a enhancement suggestion as there are monthly Play system updates that you can't receive because of root - just like monthly OS updates, so I was just wondering if adding support for updating those was something that interests you or in the scope, but if not I completely understand!

badabing2005, 2025-04-18:
No it's not in the scope of this project.
You can download them from apk mirror sites and install them manually, you do not need to unroot to do so.

closed 2025-04-18 by badabing2005