PixelUpdater/PixelUpdater #21: Pixel updater keeps freeze on "Finalizing OTA update"

kitkike, 2025-04-21:
Hello,

Thanks for this amazing module, I have tried to update my P8P with this module but it keep freeze in Finalizing OTA update for hours...

What I can do? (I send you the log)

[pu_support_2025-04-21_15-23-53.zip](https://github.com/user-attachments/files/19834532/pu_support_2025-04-21_15-23-53.zip)

badabing2005, 2025-04-21:
I don't see any errors that would give me a clue as to what is going on?
Check your current slot, reboot and see if the device switches slots, 
What Magisk are you on?

kitkike, 2025-04-21:
I also had this:

[image]

badabing2005, 2025-04-21:
Try the following.
Download clear_ota_updates from [here](https://xdaforums.com/t/pixelflasher-a-gui-tool-for-flashing-updating-rooting-managing-pixel-phones.4415453/post-87412299)
Install it as Magisk module
Reboot
Hit on the Action button
Then try to run PixelUpdater again.

kitkike, 2025-04-21:
Hello,
Can I install directly the April one without the others of January, February and march?

badabing2005, 2025-04-21:
yes

kitkike, 2025-04-21:
Hello,

I didn't know that so finally I used your pixelflasher tool with the latest global factory image (April) patched it and flashed it keeping my data.

Thank you and your job anyway, it's amazing how your tools makes our lives easier to the people who are not engineers :)

badabing2005, 2025-04-21:
Next time use OTA image instead of factory, it is safer, and if you flash it twice, you'd end up with both slots being bootable.

closed 2025-04-21 by badabing2005