PixelUpdater/PixelUpdater #28: Pixel Updater crashing when searching for update

g0r3, 2025-06-07:
Hi,

thank you for making this awesome update tool. I am using it since quite a while and it saved me so much hassle when updating my phone. Since very recently, while expecting the June update to be released, I have noticed that when I press the 'Check for updates' button, the app just crashes. Having a look in logcat, I can see that there is a `BadForegroundServiceNotificationException` happening. Not sure if that is the actual reason tho. I've attached my logcat, In case you have time to look into it, but I guess its kinda messy.  Thanks for this awesome app!

[pixelupdater.log](https://github.com/user-attachments/files/20639275/pixelupdater.log)

badabing2005, 2025-06-07:
What has changed recently?
Perhaps some background would help.
What modules you're running, what rooting mechanism are you using?

I see other errors that could be at the root of the issue, considering that it was working fine before, so it must be related to some recent change.

```
An unsupported HAL version for Keymint: "failing SetHalVersion request with error Hal(InvalidArgument, "system/keymint/ta/src/lib.rs:603: unsupported HAL version 400")".
Keyblob upgrade issues: "failing DeviceConvertStorageKeyToEphemeral request with error Hal(KeyRequiresUpgrade, "system/keymint/ta/src/lib.rs:422: keyblob with old OS patchlevel 202502 needs upgrade to current 202505")".
Errors opening the keybox on TP (Trusted Platform) by ROTLAYER2.
"failing OperationFinish request with error Hal(VerificationFailed, "system/keymint/boringssl/src/aes.rs:276: failed to finalize: ErrorStack([])")".
```

If you lost root and got root again, you might want to uninstall PixelUpdater, reboot and reinstall again, the ownership changes if you lost root and got root again.

g0r3, 2025-06-08:
Hi,

I've would have added the context, if I would have had obvious possible causes. I've last used it at beginning of May to updated my phone and everything worked fine. Now we have beginning of June, I just wanted to have a look if a new update is there and I've noticed that it is not working anymore.

The modules I've currently run:
```Anti bootloop
Integrity Box
Magisk built-in BusyBox
PixelUpdater
Play Integrity Fork
Shamiko
Systemless Hosts
Tricky Store
Zygisk Next
zygisk-detach
```

I've never lost root since using it the last time. I've changed most Play Integrity related modules since last week to gain strong integrity.

I've now de- and reinstalled PixelUpdater. The issue now  became even worse: I cannot see the Pixel Updater Icon anymore in the App Drawer, but a `com.github.pixelupdater.pixelupdater.MainAppliaction` instead (See screenshot). Please let me know what else you need

[pixelupdater2.log](https://github.com/user-attachments/files/20643409/pixelupdater2.log)

g0r3, 2025-06-13:
May I ask why this issue got marked as completed? I still have the problem, deleting and reinstalling PixelUpdater does not work for me, neither with or without reboot in between the de- and reinstallation.

Simplydat seems to have had a different problem and used my issue to comment.

badabing2005, 2025-07-07:
@g0r3 You have other issues that most likely are contributing to your problem.
No one else is having this problem, and without you addressing your other issues, it's impossible to attribute the issue being in PixelUpdater.
Start with eliminating all your root modules / tools, and then try again.

maxulloa01, 2025-11-24:
I am having the same issue, reinstall does not work.