PixelUpdater/PixelUpdater #29: SocketException (Binding socket to network 103 failed: EPERM (Operation not permitted))

CRCinAU, 2025-06-23:
When checking for new updates, PixelUpdater has recently been throwing the following error:

[image]

I tried removing the Magisk module, rebooting and reinstalling, then rebooting, but the same error occurs.

I get the notification from Magisk that SuperUser has been granted.

CRCinAU, 2025-06-24:
Interestingly, I went and did a full factory reset of the device, installed the latest Android 16 zip and this seemed to resolve the problem.

It wasn't the only thing that broke in the OTA update to Android 16 - and the other things that were also broken now work correctly too.

badabing2005, 2025-06-24:
Thanks for the update, glad it worked for you and good to know it wasn't a problem in PU.

closed 2025-06-24 by badabing2005