PixelUpdater/PixelUpdater #32: Pixel Updater app keeps crashing on qpr2 beta2

dary1zhu, 2025-09-21:
I used Pixel Updater to update my Pixel 10 Pro XL to Android 16 qpr2 beta 2. After the update the Pixel Updater app can't open and keeps crashing. I don't know if it's related to Magisk not working on qpr2 beta 2. But here's the log.

[crash-zygote64-21_09-17-06-57_368.zip](https://github.com/user-attachments/files/22450425/crash-zygote64-21_09-17-06-57_368.zip)

badabing2005, 2025-09-22:
As you stated, Magisk does not work with Android 16 qpr2 beta 2.
So I'm guessing that you're no longer rooted.

[image]

closed 2025-09-22 by badabing2005

badabing2005, 2025-09-22:
no update, it's on the back burner.

dary1zhu, 2025-10-05:
Now I'm using magisk alpha. Still crashed.

[crash-com-github-pixelupdater-pixelupdater-05_10-22-07-29_727.zip](https://github.com/user-attachments/files/22709311/crash-com-github-pixelupdater-pixelupdater-05_10-22-07-29_727.zip)

atikbinmasud, 2025-10-17:
it's still crashing even if i just install the apk.

TostoyKir, 2025-10-29:
I’ve installed this build, but unfortunately it didn’t help.
Logcat does detect the app, but no logs appear in the output when I try to record a report.
I’ve also tried with different filters (e.g. adb logcat *:E and adb logcat | grep <package_name>), but nothing shows up.
Let me know if you’d like me to test something else.
> @TostoyKir, @atikbinmasud @dary1zhu Please give this build a try.
> https://github.com/PixelUpdater/PixelUpdater/actions/runs/18893652954
> 
> Please note that I don't have QPR2 on my device, hence the changes haven't been tested on QPR2.
> 
> > i don't like this aggressive behavior, i simply said that by trying to do these operations myself i almost screwed up, it would be very cool if this issue was fixed instead of complaining "oh it never supported beta" since whenever the new QPR will release you will THEN have people complaining about crashes, so why not fix it now so that doesn't happen? it's not about kernelsu/magisk, its about the app itself crashing in the new QPR
> 
> @kekkodance You have the audacity to call my behavior aggressive? you entitled, spoiled, ungrateful, and demanding user.
> While it is true that our projects are shared freely and worked on during our personal time, this does not mean users should feel entitled to demand changes or expect immediate solutions.

badabing2005, 2025-10-29:
@TostoyKir 
Can you provide the output of the following commands?

```
cat /sys/fs/selinux/policyvers

cat /system/etc/selinux/plat_seapp_contexts

dmesg | grep -i "avc.*denied" | grep -i pixelupdater

ls -Z /dev/block/

cat /data/adb/pixelupdater/pixelupdater_selinux.log
```

TostoyKir, 2025-10-29:
Everything done on my side.

~ $ su
:/data/data/com.termux/files/home #
:/data/data/com.termux/files/home #
at /sys/fs/selinux/policyvers                                    <
at /system/etc/selinux/plat_seapp_contexts                       <_
isSystemServer=true domain=system_server_startup
user=_app seinfo=platform name=com.android.traceur domain=traceur_app type=app_data_file levelFrom=all
user=system seinfo=platform domain=system_app type=system_app_data_file
user=system seinfo=platform isPrivApp=true name=com.android.DeviceAsWebcam domain=device_as_webcam type=system_app_data_file levelFrom=all
user=bluetooth seinfo=bluetooth domain=bluetooth type=bluetooth_data_file
user=network_stack seinfo=network_stack domain=network_stack type=radio_data_file
user=nfc seinfo=platform domain=nfc type=nfc_data_file
user=nfc seinfo=nfc domain=nfc type=nfc_data_file
user=secure_element seinfo=platform domain=secure_element levelFrom=all
user=radio seinfo=platform domain=radio type=radio_data_file
user=shared_relro domain=shared_relro levelFrom=all
user=shell seinfo=platform domain=shell name=com.android.shell type=shell_data_file
user=webview_zygote seinfo=webview_zygote domain=webview_zygote
user=_isolated domain=isolated_app levelFrom=user
user=_isolated isIsolatedComputeApp=true domain=isolated_compute_app levelFrom=user
user=_sdksandbox domain=sdk_sandbox_34 type=sdk_sandbox_data_file levelFrom=all
user=_sdksandbox isSdkSandboxNext=true domain=sdk_sandbox_next type=sdk_sandbox_data_file levelFrom=all
user=_sdksandbox isSdkSandboxAudit=true domain=sdk_sandbox_audit type=sdk_sandbox_data_file levelFrom=all
user=_app seinfo=app_zygote domain=app_zygote levelFrom=user
user=_app seinfo=media domain=mediaprovider type=app_data_file levelFrom=user
user=_app seinfo=platform domain=platform_app type=app_data_file levelFrom=user
user=_app isEphemeralApp=true domain=ephemeral_app type=app_data_file levelFrom=all
user=_app isPrivApp=true domain=priv_app type=privapp_data_file levelFrom=user
user=_app isPrivApp=true name=com.google.android.permissioncontroller domain=permissioncontroller_app type=privapp_data_file levelFrom=all
user=_app seinfo=media isPrivApp=true name=com.android.providers.media.module domain=mediaprovider_app type=privapp_data_file levelFrom=all
user=_app seinfo=media isPrivApp=true name=com.android.providers.media.module:* domain=mediaprovider_app type=privapp_data_file levelFrom=all
user=_app isPrivApp=true name=com.google.android.providers.media.module domain=mediaprovider_app type=privapp_data_file levelFrom=all
user=_app isPrivApp=true name=com.google.android.providers.media.module:* domain=mediaprovider_app type=privapp_data_file levelFrom=all
user=_app seinfo=platform isPrivApp=true name=com.android.permissioncontroller domain=permissioncontroller_app type=privapp_data_file levelFrom=all
user=_app isPrivApp=true name=com.android.vzwomatrigger domain=vzwomatrigger_app type=privapp_data_file levelFrom=all
user=_app isPrivApp=true name=com.android.rkpdapp domain=rkpdapp type=privapp_data_file levelFrom=all
user=_app isPrivApp=true name=com.google.android.rkpdapp domain=rkpdapp type=privapp_data_file levelFrom=all
user=_app isPrivApp=true name=com.google.android.gms domain=gmscore_app type=privapp_data_file levelFrom=user
user=_app isPrivApp=true name=com.google.android.gms.* domain=gmscore_app type=privapp_data_file levelFrom=user
user=_app isPrivApp=true name=com.google.android.gms:* domain=gmscore_app type=privapp_data_file levelFrom=user
user=_app isPrivApp=true name=com.google.android.gsf domain=gmscore_app type=privapp_data_file levelFrom=user
user=_app minTargetSdkVersion=34 domain=untrusted_app type=app_data_file levelFrom=all
user=_app minTargetSdkVersion=32 domain=untrusted_app_32 type=app_data_file levelFrom=all
user=_app minTargetSdkVersion=30 domain=untrusted_app_30 type=app_data_file levelFrom=all
user=_app minTargetSdkVersion=29 domain=untrusted_app_29 type=app_data_file levelFrom=all
user=_app minTargetSdkVersion=28 domain=untrusted_app_27 type=app_data_file levelFrom=all
user=_app minTargetSdkVersion=26 domain=untrusted_app_27 type=app_data_file levelFrom=user
user=_app domain=untrusted_app_25 type=app_data_file levelFrom=user
user=_app minTargetSdkVersion=28 fromRunAs=true domain=runas_app levelFrom=all
user=_app fromRunAs=true domain=runas_app levelFrom=user
user=_app isPrivApp=true name=com.android.virtualization.terminal domain=vmlauncher_app type=privapp_data_file levelFrom=all
user=_app isPrivApp=true name=com.android.virtualization.terminal:* domain=vmlauncher_app type=privapp_data_file levelFrom=all
user=_app isPrivApp=true name=com.google.android.adservices.api domain=adservices type=privapp_data_file levelFrom=all
user=_app isPrivApp=true name=com.github.pixelupdater.pixelupdater domain=pixelupdater_app type=app_data_file levelFrom=all
mesg | grep -i "avc.*denied" | grep -i pixelupdater              <
mesg | grep -i "avc.*denied" | grep -i pixelupdater              <
1|:/data/data/com.termux/files/home # ls -Z /dev/block
u:object_r:block_device:s0                bootdevice  u:object_r:loop_device:s0                 loop50
u:object_r:block_device:s0                by-name     u:object_r:loop_device:s0                 loop6
u:object_r:dm_device:s0                   dm-0        u:object_r:loop_device:s0                 loop7
u:object_r:dm_device:s0                   dm-1        u:object_r:loop_device:s0                 loop8
[... cut ...]
Here is a brief report from ChatGPT 

SELinux report:

PixelUpdater has its own SELinux domain: pixelupdater_app (defined in plat_seapp_contexts).

No SELinux denials found in dmesg for PixelUpdater.

Log file /data/adb/pixelupdater/pixelupdater_selinux.log does not exist — likely no write access or daemon didn’t start.

All /dev/block/* entries have valid SELinux labels.

Most likely the PixelUpdater daemon isn’t starting or can’t write to /data/adb/pixelupdater/ due to SELinux context restrictions or missing permissions.

badabing2005, 2025-10-30:
@TostoyKir
Another try
https://github.com/PixelUpdater/PixelUpdater/actions/runs/18954544802

TostoyKir, 2025-10-31:
Unfortunately, still nothing. I’m ready to continue — tell me what to do next.
> @TostoyKir
> Another try
> https://github.com/PixelUpdater/PixelUpdater/actions/runs/18954544802

badabing2005, 2025-11-01:
@TostoyKir 
Another build
https://github.com/PixelUpdater/PixelUpdater/actions/runs/18998385429

PS: I also made a small change to the previously attached diagnose script, download a fresh copy if it needs to be run.

atikbinmasud, 2025-11-01:
> [@TostoyKir](https://github.com/TostoyKir) Another build https://github.com/PixelUpdater/PixelUpdater/actions/runs/18998385429
> 
> PS: I also made a small change to the previously attached diagnose script, download a fresh copy if it needs to be run.

not working sir.

TostoyKir, 2025-11-01:
> [@TostoyKir](https://github.com/TostoyKir) Another build https://github.com/PixelUpdater/PixelUpdater/actions/runs/18998385429
> 
> PS: I also made a small change to the previously attached diagnose script, download a fresh copy if it needs to be run.

Unfortunately, still no luck.
First I installed it over the existing version, then completely removed and did a clean install.
The result is the same — nothing loads. Judging by the logs, the app doesn’t seem to be installed at all.

[diagnose_selinux.log](https://github.com/user-attachments/files/23285802/diagnose_selinux.log)

atikbinmasud, 2025-11-09:
what's the update of this issue?

TostoyKir, 2025-12-05:
Well, everything finally works. The only thing I couldn’t test is the actual functionality — I’m waiting for QPR3 to be released. I noticed one issue: the app name on the watch is shifted; this didn’t happen before. Screenshot attached.[image]